celabshq/libcrux-iot #6: Side-channel analysis of ML-KEM & ML-DSA

franziskuskiefer, 2024-09-11:


franziskuskiefer, 2025-09-05:
@keks can you double check that we have all the notes on the doc, then we can close this one I think.

keks, 2025-11-17:
@franziskuskiefer There is this tab in the libcrux-iot doc that I did not participate in writing: https://docs.google.com/document/d/1V8RTIPkUA-eQkOH_MZ6fQkEyKWF0gXxnpL_Vll4QIm0/edit?tab=t.6tltit90ripu#heading=h.ih80qfmcclho

I do not have a good overview of what side-channel attacks there are against ML-DSA and ML-KEM, and this issue doesn't really go into much detail what is wanted here.

franziskuskiefer, 2025-11-18:
- describe secrets crate/types approach
- describe potential timing issues in mlkem and mldsa (esp. something like kyberslash2)

franziskuskiefer, 2026-03-09:
I added some text for the reports on the [doc](https://docs.google.com/document/d/1V8RTIPkUA-eQkOH_MZ6fQkEyKWF0gXxnpL_Vll4QIm0/edit?tab=t.bwh0alowkq4p). @jschneider-bensch please review and edit, then we can close this.