celabshq/libcrux-iot #14: Utility script

franziskuskiefer, 2024-09-17:
Build a utility script to easily build, test, and benchmark ML-KEM and ML-DSA for the boards in [parameters.md](https://github.com/cryspen/libcrux-iot/blob/main/parameters.md) (that we support).

keks, 2024-10-07:
Notes on how to tackle this: https://md.cryspen.com/5CY70Y0dTqqG2lZnFqzVeA#

franziskuskiefer, 2024-10-17:
@jschneider-bensch what is needed to close this issue?

jschneider-bensch, 2024-10-17:
Good question, it's indeed not well defined. We have the test & benchmark runner which runs on the boards itself and offers some convenience in writing tests and benchmarks across at least the Cortex-M boards at the moment. The runner outputs its events in a custom format via the debug probe. We should have another component on the host which can collect the runners output and processes it further, e.g. it could update our running benchmark collection table. Jan and I talked about this (but didn't make an issue of it). I'll make a sub-issue for that here, so we know when this is done?

franziskuskiefer, 2024-10-17:
👍🏻 sounds good. Let's make a sub-issue and then close this when it's done.