celabshq/libcrux-iot #41: Caching in FPU registers

jschneider-bensch, 2025-01-23:
It may be beneficial to keep $\zeta_i$ that are used in several layers cached by storing them in FPU registers, see [AHKS22](https://kannwischer.eu/papers/2022_fasterkyberdilithiumm4.pdf), section 3.1 for details.

jschneider-bensch, 2025-03-10:
To me, it remains unclear at this point, if we can exercise the necessary level of control over floating point registers to implement this.
In addition, we found out with #40 that even substantial optimizations in the arithmetic do not contribute in a major way to overall performance improvements, currently.

@franziskuskiefer Unless you think we should leave it open, I would close this as not planned at this point, okay?

(If we could demonstrate a workable style that allows this kind of optimization, that could be used to cache values elsewhere, potentially. But that's fairly speculative.)

franziskuskiefer, 2025-03-10:
Let's close it and put it on the list of things that we looked at but that don't work here for us.

closed 2025-03-10 by jschneider-bensch