celabshq/libcrux-iot #56: Sample matrix entries on the fly

jschneider-bensch, 2025-03-31:


franziskuskiefer, 2025-05-19:
Assess if we want to do this. It would require doing a separate code path because it's not useful for the unpacked version.

jschneider-bensch, 2025-05-26:
I now have a proof of concept for sampling the matrix entries on the fly in CPA encryption on branch [jonas/on-the-fly-matrix](https://github.com/cryspen/libcrux-iot/tree/jonas/on-the-fly-matrix).

The measurements are as follows:
```
Repository at commit: 112f712008187355725b044b74901543efcecaf1
Measurement: ML-KEM 768 Key Generation
  Stack begins at: 0x200a0000
  Found lowest stack address in use at: 0x2009e5fc
  Max. stack usage: 6660 Bytes

Measurement: ML-KEM 768 Encapsulation
  Stack begins at: 0x200a0000
  Found lowest stack address in use at: 0x20099f40
  Max. stack usage: 24768 Bytes

Measurement: ML-KEM 768 Decapsulation
  Stack begins at: 0x200a0000
  Found lowest stack address in use at: 0x20099b00
  Max. stack usage: 25856 Bytes
```

And for comparison, this is the "before" version, at https://github.com/cryspen/libcrux-iot/pull/84/commits/74e557b302a2d9a78d777a6f3a694843a7742463:

```
Repository at commit: b955743aca7d671a014279a44ab6e39faaf62325
Measurement: ML-KEM 768 Key Generation
  Stack begins at: 0x200a0000
  Found lowest stack address in use at: 0x2009e5fc
  Max. stack usage: 6660 Bytes

Measurement: ML-KEM 768 Encapsulation
  Stack begins at: 0x200a0000
  Found lowest stack address in use at: 0x20099358
  Max. stack usage: 27816 Bytes

Measurement: ML-KEM 768 Decapsulation
  Stack begins at: 0x200a0000
  Found lowest stack address in use at: 0x20098a7c
  Max. stack usage: 30084 Bytes
```

That's a 10% improvement for encapsulation, and 14% better for decapsulation.

Interestingly, this also had a slightly positive effect on performance, reducing cycles for encaps and decaps by around 2% each:
```

l,0,16000000,ML-KEM Benchmark
b,r,bench_keygen,0
b,d,bench_keygen,0,1774764
b,r,bench_keygen,1
b,d,bench_keygen,1,1774760
b,r,bench_keygen,2
b,d,bench_keygen,2,1774763
b,r,bench_keygen,3
b,d,bench_keygen,3,1774763
b,r,bench_keygen,4
b,d,bench_keygen,4,1774759
b,r,bench_encaps,0
b,d,bench_encaps,0,1839382
b,r,bench_encaps,1
b,d,bench_encaps,1,1839385
b,r,bench_encaps,2
b,d,bench_encaps,2,1839383
b,r,bench_encaps,3
b,d,bench_encaps,3,1839381
b,r,bench_encaps,4
b,d,bench_encaps,4,1839379
b,r,bench_decaps,0
b,d,bench_decaps,0,2022628
b,r,bench_decaps,1
b,d,bench_decaps,1,2022633
b,r,bench_decaps,2
b,d,bench_decaps,2,2022633
b,r,bench_decaps,3
b,d,bench_decaps,3,2022629
b,r,bench_decaps,4
b,d,bench_decaps,4,2022629
```

versus

```
l,0,16000000,ML-KEM Benchmark
b,r,bench_keygen,0
b,d,bench_keygen,0,1771195
b,r,bench_keygen,1
b,d,bench_keygen,1,1771196
b,r,bench_keygen,2
b,d,bench_keygen,2,1771195
b,r,bench_keygen,3
b,d,bench_keygen,3,1771200
b,r,bench_keygen,4
b,d,bench_keygen,4,1771196
b,r,bench_encaps,0
b,d,bench_encaps,0,1880683
b,r,bench_encaps,1
b,d,bench_encaps,1,1880682
b,r,bench_encaps,2
b,d,bench_encaps,2,1880684
b,r,bench_encaps,3
b,d,bench_encaps,3,1880688
b,r,bench_encaps,4
b,d,bench_encaps,4,1880679
b,r,bench_decaps,0
b,d,bench_decaps,0,2078052
b,r,bench_decaps,1
b,d,bench_decaps,1,2078051
b,r,bench_decaps,2
b,d,bench_decaps,2,2078050
b,r,bench_decaps,3
b,d,bench_decaps,3,2078054
b,r,bench_decaps,4
b,d,bench_decaps,4,2078052
```

closed 2025-06-05 by jschneider-bensch