celabshq/libcrux-iot #100: Document differences to libcrux code

franziskuskiefer, 2025-08-03:
Document differences to the libcrux code with an indicator to whether this should get proven and what may be hard about it.

jschneider-bensch, 2025-08-05:
I've summarized important difference and potential targets for (re-)verification [here](https://docs.google.com/document/d/1V8RTIPkUA-eQkOH_MZ6fQkEyKWF0gXxnpL_Vll4QIm0/edit?tab=t.5yyqpitlnt6r) in the doc.

franziskuskiefer, 2025-08-06:
Looks good. I left a few comments. You already started saying what may be interesting to verify on mldsa. Can you add a list at the bottom of the doc with things that currently look like worthwhile targets in the diff? This is of course in addition to restoring verification of all the code that's shared.

franziskuskiefer, 2025-08-06:
lgtm, let's close this and start looking at the actual pieces that we want to verify once we have extraction going again (#9).

closed 2025-08-06 by franziskuskiefer